ayufan/camera-streamer #157: Failure to get camera-streamer to work on latest Bookworm on Pi 3B - worked of box on Bullseye.

jferguson4, 2024-10-13:
Using a Module3 camera, an older 3B, and Bullseyr, the install and first successful operation using http://< ipaddress>:8080/stream worked so quickly and about as fast as I can type, I couldn't believe it.  I toggled support legacy camera off.

I spent three days wresrtling with bookworm and never got anywhere.
People who are going to use this for a dedicated installation might consider that it may be more important to get it up and running than having the latest Raspberry incarnation.

john f

tophe75, 2025-01-03:
Ok, so what I did to solve this.

There is no camera-streamer-raspi-v3-12MP.service in example folder which mean it can't be copied, so instead try: 
`sudo cp /usr/share/camera-streamer/examples/camera-streamer-generic-usb-cam.service /etc/systemd/system/camera-streamer.service`
`systemctl enable camera-streamer`
`systemctl start camera-streamer`

No I can see the stream @ http://< ipaddress>:8080/stream

fabnavigator, 2025-01-21:
I am also trying to get this working with Bookworm on a Pi 3B. Here is my camera:

$ rpicam-hello --list-cameras

Available cameras
-----------------
0 : imx219 [3280x2464 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [83.70 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]

I also used the camera-streamer-generic-usb-cam.service example.

When I try:

$ camera-streamer --camera-path=/dev/video0 --http-listen=0.0.0.0 --http-port=8080

I get:

camera-streamer Version: 0.2.8 (bc23191)
util/http/http.c: ?: HTTP listening on 0.0.0.0:8080.
device/v4l2/device.c: CAMERA: Device path=/dev/video0 fd=14 opened
device/v4l2/device_options.c: CAMERA: Configuring option 'Horizontal Flip' (00980914) = 0
device/v4l2/device_options.c: CAMERA: Configuring option 'Vertical Flip' (00980915) = 0
device/buffer_list.c: CAMERA:capture: Using: 1920x1080/YUYV, buffers=3, bytesperline=3840, sizeimage=4.0MiB
device/buffer_list.c: CAMERA:capture: Opened 3 buffers. Memory used: 11.9 MiB
device/v4l2/device.c: SNAPSHOT: Device path=/dev/video31 fd=19 opened
device/v4l2/buffer_list.c: SNAPSHOT:output:mplane: Requested resolution=1920x1080 is unavailable. Got 1920x1088.
device/buffer_list.c: SNAPSHOT:output:mplane: Using: 1920x1056/YUYV, buffers=3, bytesperline=3840, sizeimage=3.9MiB
device/buffer_list.c: SNAPSHOT:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Using: 1920x1056/JPEG, buffers=3, bytesperline=0, sizeimage=4.0MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Opened 3 buffers. Memory used: 12.0 MiB
device/v4l2/device.c: CAMERA: ioctl(ret=-1, errno=25): Can't set FPS
device/device.c: CAMERA: Setting frame interval_us=33333 for FPS=30
device/buffer_list.c: CAMERA:capture: Streaming started... Was 0 of 3 enqueud
device/v4l2/device_options.c: CAMERA: The 'AfTrigger=1' was failed to find.
device/v4l2/device_options.c: SNAPSHOT: Configuring option 'Compression Quality' (009d0903) = 80
device/links.c: ?: Link 0: CAMERA:capture[1920x1080/YUYV/3] => [SNAPSHOT:output:mplane[1920x1056/YUYV/3]]
device/links.c: ?: Link 1: SNAPSHOT:capture:mplane[1920x1056/JPEG/3] => [SNAPSHOT-CAPTURE, STREAM-CAPTURE]
device/buffer_list.c: SNAPSHOT:output:mplane: Streaming started... Was 0 of 3 enqueud
device/buffer_list.c: SNAPSHOT:capture:mplane: Streaming started... Was 0 of 3 enqueud

When I try to access the stream I get "server error" and see this on the Pi:

util/http/http.c: HTTP8080/8: Request 'GET' '/stream' ''
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
util/http/http.c: HTTP8080/8: Client disconnected 192.168.0.15.

Any help will be appreciated.