ayufan/camera-streamer #158: Raspberry Pi - WebRTC Stream Not Available

SCH00N3R, 2024-10-28:
Hello,

I followed your instructions on installing camera-streamer to my Raspberry Pi. I can connect to the web interface and can see the /stream working. However, when I click on /webrtc I get a blank playback screen with an infinitely spinning wheel. Also, if I go to the control section, The Start WebRTC button is greyed out. What did I miss?

Thank you!

SCH00N3R

stablestud, 2025-01-12:
Same for me. On what RasbianOS version are you?
`cat /etc/os-release`

macavelly, 2025-03-02:
Did you ever figure this out? Maybe its just me but the instructions wasn't very clear how to enable webrtc.

"The WebRTC is accessible via http://<ip>:8080/webrtc by default and is available when there's H264 output generated."

I'm using the precompiled binary and my service file looks like this:

`[Unit]
Description=camera-streamer web camera for USB camera on Generic platform
After=network.target

[Service]
ExecStart=/usr/bin/camera-streamer \
  -camera-path=/dev/video0 \
  -camera-format=JPEG \
  -camera-width=1280 -camera-height=720 \
  -camera-fps=15 \
  ; use two memory buffers to optimise usage
  -camera-nbufs=3 \
  -camera-video.height=720 \
  -camera-snapshot.height=720 \
  -camera-stream.height=720 \
  --http-listen=0.0.0.0 \
  --http-port=8080

DynamicUser=yes
SupplementaryGroups=video i2c
Restart=always
RestartSec=10
Nice=10
IOSchedulingClass=idle
IOSchedulingPriority=7
CPUWeight=20
AllowedCPUs=1-2
MemoryMax=250M

[Install]
WantedBy=multi-user.target`

But when I click on the webrtc link, just the spinning loading bar.

WereCatf, 2025-08-26:
Use `--webrtc-disable_client_ice=1`

MartinEmrich, 2025-09-03:
Same issue here, `--webrtc-disable_client_ice=1` has no effect.

MartinEmrich, 2025-09-03:
... and the solution for me was adding `gpu_mem=128` to the `config.txt`.

This IS documented, but without it the normal MJPEG streams still work, and there are no error messages whatsoever. So I noticed it just by chance.